Leathong/openscad-LSP #2: Allow specifying port

Lenbok, 2022-05-07:
It appears that the port is hardcoded to 3245, which would prevent running multiple instances on the same machine. Most other tcp servers allow the port to be specified, e.g. via command line flag `--port`. Can this be added?

Lenbok, 2022-05-07:
BTW, because of the above I couldn't yet work out how to connect to emacs lsp-mode, as the examples I could see wanted to dictate the port. 

However, I simply swapped the main of your server implementation in `main.rs` to use the stdio-based connection that dzhu's implementation used, and I can now use your server inside emacs lsp-mode.  Why did you choose to implement tcp over a stdio based connection?

Leathong, 2022-05-07:
> It appears that the port is hardcoded to 3245, which would prevent running multiple instances on the same machine. Most other tcp servers allow the port to be specified, e.g. via command line flag `--port`. Can this be added?

Sure, it's a nice feature. I will add it in the next few days.

Leathong, 2022-05-07:
> 

> 

It may be need to output some information via stdio in the future.

Leathong, 2022-05-07:
> BTW, because of the above I couldn't yet work out how to connect to emacs lsp-mode, as the examples I could see wanted to dictate the port.
>
> However, I simply swapped the main of your server [...]

It may be need to output some information via stdio in the future.

Leathong, 2022-05-08:
done

closed 2022-05-08 by Leathong

Lenbok, 2022-05-08:
Thanks, the tcp connection method is working for me now.