Leathong/openscad-LSP #4: RFE Support returning function/module documentation

Lenbok, 2022-05-09:
I don't know what the underlying LSP service is, but several other language servers support returning documentation for functions and variables I assume based on the presence of comments in the corresponding source. It would be fantastic if you could support that as well.

The difficulty I can see is that there isn't really a standard for documenting openscad modules (although if you were to implement support for some documentation method that may very well help to establish a standard). Most libraries include module documentation in the comment immediately preceding the module definition, but the actual format can vary e.g.:

BOSL: https://github.com/revarbat/BOSL2/blob/master/beziers.scad#L178-L209
MCAD uses JavaDoc style: https://github.com/SolidCode/MCAD/blob/master/triangles.scad#L27-L34

I tend to use Javadoc style myself. I have seen a couple of libraries that include module documentation immediately after the module declaration, but this seems quite rare.

Leathong, 2022-05-09:
This is a useful feature, I will try in the near future.

Leathong, 2022-05-11:
Already added a simple support for that.

closed 2022-05-11 by Leathong