Leathong/openscad-LSP #5: How send configuration?

Lenbok, 2022-05-24:
I see you have added some support for formatting and configuration settings, but I don't know how to get the settings to work (I wanted to customize the formatting style, since my clang-format doesn't include a "Microsoft" style I wanted to use a style equivalent to `--style '{BasedOnStyle: llvm, SpacesInContainerLiterals: false}'`.

I took a stab at getting emacs' lsp-mode to send configuration settings, and from the trace it is sending the following json:
```
[Trace - 06:19:32 PM] Sending notification 'workspace/didChangeConfiguration'.
Params: {
  "settings": {
    "openscad": {
      "searchPaths": "",
      "fmt_exe": "clang-format",
      "fmt_style": "llvm"
    }
  }
}
```
But the server is outputting `[error] missing field 'searchPaths'`. 

So my questions:
- What should the transmitted configuration  json look like?
- Do all parameters have to be provided or can some be left out to retain the default?

Lenbok, 2022-05-24:
Should your impl be looking for the settings to be within a section, like e.g.: https://github.com/denoland/deno/blob/main/cli/lsp/language_server.rs#L924

Lenbok, 2022-05-24:
I managed to send configuration settings by hacking the json to remove the "openscad" section. I think this server should expect it's settings within a section as other language servers do, so I'm leaving this open.

Leathong, 2022-05-24:
done with this commit https://github.com/Leathong/openscad-LSP/commit/a3141af268ef7ea385edc2d2068ad0777ab0bfdd

closed 2022-05-24 by Leathong

Lenbok, 2022-05-24:
It does not crash if I send settings as:
```
[Trace - 08:19:16 AM] Sending notification 'workspace/didChangeConfiguration'.
Params: {
  "settings": {
    "openscad": {
      "search_paths": "",
      "fmt_exe": "clang-format",
      "fmt_style": ".clang-format"
    }
  }
}
```
But it makes sense for the server to be robust to keys that do not get sent. (I am testing 0f9079d)