Leathong/openscad-LSP #11: Occasional crash

Lenbok, 2022-07-11:
This is built from master at [6c1579c](https://github.com/Leathong/openscad-LSP/commit/6c1579c52223ce1c291339cac2ee365de51d6230)

```
thread 'main' panicked at 'called `Option::unwrap()` on a `None` value', src/main.rs:414:37
```

Haven't yet worked out to reproduce it.

Leathong, 2022-07-11:
It looks like IDE send a did_change_text_document notification with missing range.

Lenbok, 2022-07-11:
I guess so - I turned on tracing and this is what appears in the log:
```
[Trace - 10:17:03 PM] Sending notification 'textDocument/didChange'.
Params: {
  "textDocument": {
    "uri": "file:///home2/len/foo/foo.scad",
    "version": 1
  },
  "contentChanges": [
    {
      "text": "\nfunction bar(x) = x * x;\n\nmodule bar() {\n  bAr = bar(2);\n  translate([0, 0, 0]) circle(r = 10);\n}\n\nuse<bar.scad>\n"
    }
  ]
}
```

It would be nice if the server didn't crash though :-)

Lenbok, 2022-07-11:
BTW, it sent the entirety of the document in the text parameter, so maybe range is optional in that case?

MaxVerevkin, 2022-07-11:
@Lenbok which text editor/IDE do you use?

Lenbok, 2022-07-11:
I am using emacs with lsp-mode. According to the lsp specs "An event describing a change to a text document. If only a text is provided it is considered to be the full content of the document.", so it seems a valid thing to do. (I don't know under what conditions emacs lsp-mode decides to do whole content vs ranged sections).

Leathong, 2022-07-12:
@Lenbok Try this https://github.com/Leathong/openscad-LSP/commit/6ead5f8452ab8a63f723779d398889763ed3eb6d.

closed 2022-07-12 by Lenbok