Leathong/openscad-LSP #14: Error when using 'Preview in OpenSCAD'

Neutralization, 2022-11-08:
#### vscode settings:
```
{
    "scad-lsp.fmtExePath": "D:\\clang-format.exe",
    "scad-lsp.defaultParam": true,
    "scad-lsp.launchPath": "C:\\Users\\username\\scoop\\apps\\openscad\\2021.01\\openscad.exe",
    "scad-lsp.fmtStyle": "Microsoft",
}
```

When using 'Preview in OpenSCAD':
```
terminate called after throwing an instance of 'boost::filesystem::filesystem_error'
  what():  boost::filesystem::read_symlink: ��֧�ָ�����: "C:"
```

I wonder if it's scoop's problem and normal installed openscad to ```"C:\\Program Files\\OpenSCAD\\openscad.exe"``` and changed launchpath, still same error.

using vscode 1.73.0 on windows 10 19045
clang-format version 15.0.3

Leathong, 2022-11-09:
I don't have a Windows computer around right now, i will check it on weekends.

Leathong, 2022-11-09:
You can try to install to 'C:\\Program Files\\Openscad\\openscad.exe', and delete the 'lanchpath'

Neutralization, 2022-11-09:
> You can try to install to 'C:\Program Files\Openscad\openscad.exe', and delete the 'lanchpath'

Still got same error. I'm using OpenSCAD 2021.01
Then I found OpenSCAD has [Development Snapshots](https://openscad.org/downloads#snapshots) with 2022.11.07 and it works. So it's totally OpenSCAD's problem I guess.

Leathong, 2022-11-10:
OK，I‘ll close this issue, feel free to commit new one if have any problem.

closed 2022-11-10 by Leathong