Leathong/openscad-LSP #35: Update `tree-sitter-openscad`

mkatychev, 2025-02-05:
In a prelude to https://github.com/Leathong/openscad-LSP/pull/33 I've added many more grammar cases to the [`tree-sitter-openscad` grammar](https://github.com/bollian/tree-sitter-openscad/pull/18#issuecomment-2608385509) however the repo is [possibly unmaintained](https://github.com/tweag/topiary/pull/867#issuecomment-2637682210). 

The changes needed to add the edge cases that are correct in the [openscad AST](https://github.com/openscad/openscad/blob/34c0e2af798b6d059bdfd4f710e67efd7444a5ef/src/core/AST.cc#L2) would required a newer version of the [tree-sitter-openscad crates.io](https://docs.rs/tree-sitter-openscad/) crate that I have no control over.

In the meantime, adding the [coverage from the proposed 0.6 changes](https://github.com/mkatychev/tree-sitter-openscad/commits/master/test/corpus) would required the dep below to change to the git dependency or a crates.io fork:
https://github.com/Leathong/openscad-LSP/blob/e4ac57eeeb1579bfe670f25ae6de695b5608cae1/Cargo.toml#L20
* `tree-sitter-openscad.git = https://github.com/mkatychev/tree-sitter-openscad` 
* `tree-sitter-openscad = { package = "tree-sitter-openscad-fork", version = "0.6" }`
* [prefetch the grammar as a dylib](https://github.com/tweag/topiary/blob/c144f48d25d94f9b727f76356e204d08a62e90bc/topiary-config/src/lib.rs#L142-L173)

The git dependency change means that a newer version of `openscad-lsp` won't be able to be published to crates.io but a fork may also be undesirable. Option #3 requires the most work and I'm not for it, just providing an alternative. 

Let me know your throughts @Leathong

Leathong, 2025-02-06:
Your pull request for tree-sitter-openscad contains too many changes, making it difficult to review. It would be better to break it down into several smaller PRs. Additionally, it seems some editor config files and symbol files have been included in the commit, should these be removed?

Leathong, 2025-02-06:
If the maintainer has not merged your code for a long time, I think it is ok to use your repository directly.

Lenbok, 2025-02-06:
Maybe you could ask the openscad org if they work like to adopt your fork into the org, and then it could become the official repository for the treesitter grammar. It seems a fairly foundational library that makes sense to have in the org to ensure it stays maintained.

mkatychev, 2025-02-06:
@Lenbok who/what would be the point of contact for this?

mkatychev, 2025-02-06:
@kintel do you have any thoughts on the matter for @Lenbok's suggestion? There's some changes (more context in the issue desc.) for syntax highlighting, formatting, and LSP error messaging held back by lack of movement in the current [tree-sitter-openscad repo](https://github.com/bollian/tree-sitter-openscad), would the tree-sitter grammar do well to live under github.com/openscad?

Perhaps [github.com/tree-sitter-grammars](https://github.com/tree-sitter-grammars) is also a good location for the grammar to move to.

kintel, 2025-02-06:
I'm not familiar with tree-sitter, but we could easily host an openscad-specific grammar under the OpenSCAD github org. Someone would need to care for the repo of course. Let me know what I can do to facilitate that.

mkatychev, 2025-02-06:
Hi @kintel, thanks for the quick response. Most of the relevant changes were under my fork of tree-sitter-openscad here:
https://github.com/mkatychev/tree-sitter-openscad 

> I'm not familiar with tree-sitter

There's a quick intro [here](https://tree-sitter.github.io/tree-sitter/index.html) but tree-sitter essentially allows a lot of operations to be done on the grammar of the language just through the AST including syntax highlighting (many of grammars on github [use tree-sitter for highlighting](https://github.com/github-linguist/linguist/discussions/5572#discussioncomment-1355712)):
https://tree-sitter.github.io/tree-sitter/index.html

Lenbok, 2025-02-07:
I've sent @bollian an email directly in case he's not getting GitHub notifications to see if he's open to the idea of transferring the original repo. Let's see if there's any response in the next few days.

kintel, 2025-02-07:
Sounds good, give me the word, and I can initiate a hard fork/transfer from whatever source we want to use.

Lenbok, 2025-02-16:
There's been no reply. I suggest that @mkatychev work with @kintel to set up the hard fork into the openscad org along with whatever other changes need to happen so that the crate can get published in a way that downstream packages can start using it.

mkatychev, 2025-02-16:
> Your pull request for tree-sitter-openscad contains too many changes, making it difficult to review. It would be better to break it down into several smaller PRs. Additionally, it seems some editor config files and symbol files have been included in the commit, should these be removed?

@Leathong 
Agreed, I had one PR that was an easy to review change, however it did not make sense to make further PRs if the code is not going to get reviewed.

mkatychev, 2025-02-17:
New repo has been made here: https://github.com/openscad/tree-sitter-openscad/ @kintel  has been a huge help!

Leathong, 2025-05-12:
@mkatychev I've adapted the new dependency tree-sitter-openscad-ng, Is the new formatter ready?

mkatychev, 2025-05-12:
Hi, @Leathong the new formatter has been working since February, new edge case fix was merged just yesterday:
https://github.com/tweag/topiary/pull/972

I have not gotten around to adding it here but `topiary fmt` is what you want to call:

```
Format inputs

Usage: topiary format [OPTIONS] <--language <LANGUAGE>|FILES>

Arguments:
  [FILES]...
          Input files and directories (omit to read from stdin)
          
          Language detection and query selection is automatic, mapped from file extensions defined in the Topiary configuration.

Options:
  -t, --tolerate-parsing-errors
          Consume as much as possible in the presence of parsing errors

  -s, --skip-idempotence
          Do not check that formatting twice gives the same output

  -l, --language <LANGUAGE>
          Topiary language identifier (when formatting stdin)

  -q, --query <QUERY>
          Topiary query file override (when formatting stdin)

  -L, --follow-symlinks
          Follow symlinks (when formatting files)

  -C, --configuration <CONFIGURATION>
          Configuration file
          
          [env: TOPIARY_CONFIG_FILE]

  -M, --merge-configuration
          Enable merging for configuration files

  -v, --verbose...
          Logging verbosity (increased per occurrence)

  -h, --help
          Print help (see a summary with '-h')
```

`topiary --language openscad` should be all that is needed to format stdin.

I'm a bit tied up this month so anyone is welcome to take a crack at it.

Leathong, 2025-05-13:
@mkatychev Great，I will proceed with the remaining tasks.

closed 2025-05-14 by Leathong

Leathong, 2025-05-14:
@mkatychev Is there a way to configure the formatter? For instance, can I modify settings like indent width?

mkatychev, 2025-05-14:
@Leathong  yes you can, you need to provide a `--merge-configuration` or `--config` (to override). this is the indent value you can pass in: 
https://github.com/tweag/topiary/blob/d0d544889bf7f523a7c4d5f5bf0badd34948e2a2/topiary-config/languages.ncl#L94

The openscad one is not specified so uses default spacing of 2 (I think). We can make a PR to pass in memory but for now you can pass in a file like this to `--merge-configuration`:
```ncl
{
  languages.openscad.indent = "    ", # 4 spaces
}
```