Leathong/openscad-LSP #36: Guide or docs to get this working in lazy nvim?

skela, 2025-02-16:
Tried a few things, like adding "openscad-lsp" to Mason (which failed), manually installing it using cargo (which worked),
What magic snippet do I add to the "neovim/nvim-lspconfig" to get this working? 

Didnt have any luck with what was mentioned here, or in the lspconfig repos:

[image]

skela, 2025-02-16:
[image]

Leathong, 2025-02-17:
Sorry, I am not familiar with nvim.

jcorbin, 2025-02-17:
So the nvim-lspconfig already ships config for openscad-lsp, you don't need to define a custom server config: https://github.com/neovim/nvim-lspconfig/blob/master/doc/configs.md#openscad_lsp

I don't use the lazy distro of neovim, but here's my config for nvim + openscad lsp : https://github.com/jcorbin/home/blob/main/.config/nvim/init.lua#L483

The only custom config part aside from just "use openscad lsp" is to disable the autoformatting, since imo, there's no such thing as a sane auto formatter for openscad yet... and I dearly wish someone would prove me wrong there ;)

skela, 2025-02-17:
thx for the tips.
after trying it out on another machine, and having everything work pretty much out of the box, i figured out my issue, it failed to install via Mason due to the wrong rust toolchain being used on my machine that had issues.

have been doing some rust programming on nightly builds so prob not a very common scenario, 
running `rustup default stable` and then retrying the installation via Mason installed everything correctly. 

Now i can edit my openscad files in nvim ! 👍🏻 

Thanks for making this awesome lsp!

closed 2025-02-17 by skela