Leathong/openscad-LSP #39: Way to log debug messages?

roel-v, 2025-02-19:
I'm trying to get this language server to work with vim, using https://github.com/yegappan/lsp . I've configured it to send workspace/didChangeConfiguration with "search_paths" set to the directory where I have BOSL2 installed. Now while I do get some completions in vim, the BOSL2 modules don't show up, nor do any modules from *.scad files in the same directory as the file I'm editing.

I'm wondering if there's a way to enable logging in openscad-LSP so that I can see what config values are being set, and if they're being picked up in the right way by openscad-LSP. Right now I have no idea what is happening, or which .scad files are being scanned for module definitions. I've build openscad-LSP as a Github Action because I don't have a Rust toolchain on my machine, and I'm really trying to avoid having to install everything on my machine so that I can build logging-enabled executables... Any other tips on how the directory scanning is supposed to work (e.g., does it scan recursively?) or other troubleshooting tips? Thanks.

roel-v, 2025-02-19:
OK I've found that I had at first misconfigured my vim lsp plugin, but after fixing that, I'm still not getting completions. I've bitten the bullet and installed Rust anyway so that I could build an executable with debugging trace statements, and I can see that Server::extend_libs() is called with the right directory, but now I'm not sure any more where to look further - shouldn't having this directory set up work automatically to find symbols from .scad files in response to textDocument/completion messages?

Leathong, 2025-02-20:
The log will be output to stderror, and you should be able to see it in the console, or somewhere can check stderror and stdout. If it cannot be set correctly during runtime, I suggest passing the search path as a startup args to the server.

closed 2025-04-23 by Leathong