Leathong/openscad-LSP #41: Inconsistencies Between GitHub Source and Crates.io Package for openscad-lsp

GregoryLand, 2025-04-28:
I pulled the [AUR package](https://aur.archlinux.org/packages/openscad-lsp) for this project and attempted to update it from 1.2.5 to 1.2.6. However, the build immediately failed, reporting that the 1.2.6 version didn't exist at crates.io.

Digging deeper, I noticed that the AUR package is pulling directly from [crates.io](https://static.crates.io/crates/openscad-lsp/openscad-lsp-1.2.5.crate) instead of a GitHub release tarball.

This raised some concerns:

1. There is no GitHub tag or release for v1.2.5 to verify against crates.io contents.
2. Version v1.2.5.1 appears never to have been published to crates.io.
3. Version v1.2.6 also doesn't seem to be published to crates.io.
4. Is the crates.io package officially published and maintained by this project? If not, an unauthorized party may have uploaded the crate.
5. There does not appear to be a Github Action or any automated process publishing the crate as part of tagged releases.

Further investigation shows that the crates.io cargo archive, for openscad-lsp (v1.2.5), differs substantially from the corresponding GitHub repository:

- Several features (such as rename support) are completely missing in the crates.io package.
- The dependency tree-sitter-traversal was removed from the crates.io package, alongside all associated code.
- Introduced bugs ("defualt" typo) are present in the crates.io code but not in GitHub.

This means the published crate is not verifiably equivalent to the GitHub source code. Can you please confirm whether the crates.io package is officially maintained by this project?

Owner of crate check:
```
cargo owner --list openscad-lsp
    Updating crates.io index
Leathong (Leathong)
```

nick42d, 2025-04-28:
Following - I am the maintainer of the AUR package. The readme lists `cargo install` as an official install method, so I simply repackaged that method.

Will keep an eye on this thread and update the package if @Leathong would prefer to no longer maintain a crates.io release (or alternatively, would be happy to help with set up of github actions release - I have used `release-plz` in the past for this).

Leathong, 2025-04-28:
I didn't consistently sync GitHub updates to crates.io. I've just published the latest version to crates.io now.

GregoryLand, 2025-04-28:
Hey @Leathong, thanks for getting the update pushed to crates.io!

Since @nick42d offered to help set up GitHub Actions for automating releases, it might be a good opportunity to make the whole release process a lot easier going forward.
Would definitely help avoid sync issues between GitHub and crates.io in the future too.

Appreciate you digging into all this — thanks again!

Leathong, 2025-04-28:
Thank you for the suggestion，I'll create an Action to automate the publishing process.

nick42d, 2025-04-28:
Awesome - thanks @Leathong for the quick turnaround and awesome crate!

Here's an example release-plz action from one of my packages if it helps https://github.com/nick42d/youtui/blob/main/.github/workflows/release-plz.yml.

Feel free to reach out via email if I can help in any way (email listed on AUR package).

Cheers Nick

closed 2025-05-14 by Leathong