Leathong/openscad-LSP #47: Parser fails on floats ending with a period.

lgarron, 2025-05-15:
I have some code as follows:

```
pA = [ 0., 0. ];
pB = [ 0.5, 0.866025 ];
pC = [ 1., 0. ];
```

I wouldn't write such code on purpose, but this was the output of the previous formatter and OpenSCAD accepts it.

I think it would be appropriate to convert e.g. `0.` to either `0`, `0.`, or `0.0`, but if OpenSCAD accepts it then I think it should pass through the formatter successfully.

lgarron, 2025-05-15:
Also note that this syntax error completely hangs https://github.com/Leathong/openscad-support-vscode and prevents the file from being saved.

Leathong, 2025-05-15:
This is not a common coding style, The syntax parser is not officially developed, so it cannot guarantee 100% compatibility with all corner cases.The new formatter is implemented based on syntax parsing, so it can't work if syntax parsing failed.

mkatychev, 2025-05-29:
@lgarron  can you open an issue in https://github.com/openscad/tree-sitter-openscad/ please?

closed 2025-05-30 by Leathong