Leathong/openscad-LSP #48: Incorrect line breaks when using `echo(…)` and or coordinate access

lgarron, 2025-05-15:
I have the following code:

```scad
echo(bisector_angle_offset=bisector_angle_offset);
prev_angle = atan2((prev - point).y, (prev - point).x);
echo(prev_angle=prev_angle);
```

This is formatted as:

```scad
echo(bisector_angle_offset=bisector_angle_offset); prev_angle = atan2((prev - point) .y, (prev - point) .x);
echo(prev_angle=prev_angle);
```

I'm not sure exactly what's causing it, but it definitely looks to me like this shouldn't be intended.

Leathong, 2025-05-15:
@mkatychev Do you have any idea about this ?

mkatychev, 2025-05-15:
🤔 I can reproduce it.

mkatychev, 2025-05-15:
@Leathong fix is out: https://github.com/tweag/topiary/pull/992 while waiting for upstream merge/release, you can call `topiary format --query <openscad-format-query.scm>` to get it in as a hotfix now.

This is the file you'd have to put somewhere for the custom query:
https://github.com/tweag/topiary/blob/7046eff60e9c19171f71f6c028ebb12d19f0ccd6/topiary-queries/queries/openscad.scm

Leathong, 2025-05-16:
@lgarron Please download the newest version and try again.

closed 2025-05-16 by Leathong

lgarron, 2025-05-16:
> [@lgarron](https://github.com/lgarron) Please download the newest version and try again.

Hmm, I'm still seeing the same issue. Am I missing something?

- `leathong.openscad-language-support`: 1.3.0
- `topiary-cli`: 0.6.0 (downloaded from https://github.com/Leathong/openscad-LSP/releases/tag/v1.3.0)

Leathong, 2025-05-16:
@lgarron I can't reproduce this issue anymore. Could you delete the old one and try downloading it again?