Leathong/openscad-LSP #49: Upgraded to the new openscad lsp and topiary doesn't work

pinkfish, 2025-05-16:
Upgraded to the new version, installed topiary, updated the exe path and tried without, but formating doesn't seem to do anything now.  It was working with clang, but I see no output anywhere at all now to say it is even trying to run anything.  Is there some debug steps on how to get this working?

pinkfish, 2025-05-17:
Figured this out, looks like if topiary fails it silently fails.  It has an issue with block comments at the top of the file by the looks.

Leathong, 2025-05-17:
Could you share the code snippet that caused the syntax parsing to fail?

pinkfish, 2025-05-22:
Hi there. I did on the topiary site and they closed it there as fixed.  Thanks!  It would be nice if it didn't silently fail though.

closed 2025-05-28 by Leathong