Leathong/openscad-LSP #50: LSP hangs fails idempotence check due to comments that resemble source code

lgarron, 2025-05-18:
Repro

```shell
curl -O https://raw.githubusercontent.com/lgarron/scad/21b8084acb0b2d26b8cbc76b655d23f479775d26/variants.scad
# Using `topiary` from [text](https://github.com/Leathong/openscad-LSP/releases/tag/v1.3.0)
topiary format variants.scad
```

It looks like this is trying to indent inconsistently.

Output:

```
[2025-05-18T10:06:21Z ERROR topiary_core] Failed idempotence check
[2025-05-18T10:06:21Z ERROR topiary_core] Diff < left / right > :
     /*
    < 
    < // Allows specifying multiple variants to be rendered by `openscad-auto`: https://github.com/lgarron/dotfiles/blob/main/scripts/app-tools/openscad-auto.rs
    < //
    < // Each variant is a string, which must defined as an entry in a global BOSL2 struct named `VARIANT_DATA`.
    < //
    < // - The struct must contain an entry with the key `"default"`.
    < // - Keys are used in file names.
    < //   - They must not contain the character `.`, as this is used to compose variants (see below).
    < //   - Keys are used to create file names, so it is worth using names that will be as clear as possible in this context.
    < // - The value of each entry is a list containing:
    < //   - A required first item with a BOSL2 struct specifying a mapping of parameter names to values.
    < //   - An optional second item containing a list of parent variants to inherit from.
    < //     - Later entries override earlier entries.
    < //
    < // Each variant falls back to the parameter value from the `"default"` entry if it is not specified among the ancestors of a given entry.
    < // Note that the top-level order of entries has no effect on the parameter calculations. Parameter inheritance for a variant depends solely on the ordering of its parents (and their parents, recursively).
    < //
    < // The variant it specified in a global `VARIANT` var.
    < //
    < // - Multiple variants may be composed by specifying them with a `.` inbetween.
    < //   - Note that such variants can be defined explicitly in `VARIANT_DATA` by specifying those variants as parents. However, the compositional approach is a very convenient mechanism to avoid boilerplate.
    < //
    < // Parameters are computed by calling `get_parameter(parameter_name)` after `VARIANT` and `VARIANT_DATA` have been defined.
    < //
    < // Example usage:
    < 
    < VARIANT = "default"; // ["default", "no-version", "mini", "mini.normal-height"]
    < 
    < VARIANT_DATA = [
    <   [
    <     "default",
    <     [
    <       [
    <         ["TOTAL_HEIGHT", 5],
    <         ["TRIANGLE_EDGE_LENGTH", 25],
    <         ["INCLUDE_VERSION_ENGRAVING", true],
    <       ],
    <     ],
    <   ],
    <   [
    <     "no-version",
    <     [
    <       [
    <         ["INCLUDE_VERSION_ENGRAVING", false],
    <       ],
    <     ],
    <   ],
    <   [
    <     "mini",
    <     [
    <       [
    <         ["TOTAL_HEIGHT", 3],
    <         ["TRIANGLE_EDGE_LENGTH", 15],
    <       ],
[... 143 lines cut ...]
         let (maybe_value = __variants__maybe_get_parameter_for_variant(parameter_name, last(variant_ids), variant_ids_in_recursion_stack_for_cycle_detection)) is_undef(maybe_value) ? __variants__maybe_get_parameter_for_variants(parameter_name, list_head(variant_ids), variant_ids_in_recursion_stack_for_cycle_detection) : maybe_value
       );
     
     // export function get_parameter(parameter_name: ParameterName): ParameterValue;
     function get_parameter(parameter_name) =
       assert(!is_undef(VARIANT), "The global variable `VARIANT` must be set before calling `get_parameter(…)`") // TODO: doesn't work
       assert(!is_undef(VARIANT_DATA), "The global variable `VARIANT_DATA` must be set before calling `get_parameter(…)`") // TODO: doesn't work
       let (default_entry = struct_val(VARIANT_DATA, "default")) assert(!is_undef(default_entry), "An entry for the variant `\"default\"` must be present in `VARIANT_DATA`/")
       let (default_entry_value_for_parameter = struct_val(__variants__variant_parameters(default_entry), parameter_name)) assert(!is_undef(default_entry_value_for_parameter), str("Parameter is missing from the `\"default\"` entry in `VARIANT_DATA` (all parameters must be defined on the default variant): ", parameter_name))
       let (
         variant_ids = str_split(VARIANT, "."),
         maybe_value = __variants__maybe_get_parameter_for_variants(parameter_name, concat(["default"], variant_ids), []),
       ) let (value = is_undef(maybe_value) ? __variants__maybe_get_parameter_for_variant(parameter_name, "default", []) : maybe_value) assert(!is_undef(value), str("Unable to get parameter: ", parameter_name)) // Technically we're doing unnecessary string concatentation in the "happy" path, but this should be cheap enough.
       value;
     
    
[2025-05-18T10:06:21Z ERROR topiary] The formatter did not produce the same
    result when invoked twice (idempotence check).
    
    If this happened with the built-in query files, it is a bug. It would be
    helpful if you logged this error at
    https://github.com/tweag/topiary/issues/new?assignees=&labels=type%3A+bug&template=bug_report.md
```

The last part of the error says to file an issue at https://github.com/tweag/topiary and I can do taht if it's more appropriate.

Leathong, 2025-05-18:
It's a bit strange to hang. If the formatting fails, the server will return an error to the client.

Leathong, 2025-05-18:
@lgarron Could you share the original code?

mkatychev, 2025-05-18:
@Leathong taking a look at nls (nickel language server) which uses topiary for formatting it looks like the idempotency check is turned off:
https://github.com/tweag/nickel/blob/cd5e68544e8158e47bcdfae89c13b3e279bef1e2/core/src/format.rs#L36-L39.

@lgarron this is a good place to handle the issue since it helps track lps behaviour as well as formatting issue, thanks for finding these issues, and thanks for your patience!

mkatychev, 2025-05-19:
@lgarron do you think this is related? https://github.com/tweag/topiary/issues/997

lgarron, 2025-05-20:
> [@lgarron](https://github.com/lgarron) do you think this is related? [tweag/topiary#997](https://github.com/tweag/topiary/issues/997)

Looks quite plausible, seems it's also about an extra space of indentation in a comment!

closed 2025-05-28 by Leathong