Leathong/openscad-LSP #51: Formatting Issues with Topiary

CameronBrooks11, 2025-05-18:
I've encountered a couple of issues using Topiary to format OpenSCAD code with openscad-LSP:

### 1. Block Comments at the Start of a File Cause a Hang (Minor)

When a file begins with a block comment using `/** ... */`, formatting fails the idempotence check and hangs. The formatter tries to convert it into a regular block comment (`/* ... */`) and strips the leading asterisks from each line.

This behavior goes against common conventions — `/** ... */` is typically used for structured documentation comments (like Javadoc or Doxygen). Even though OpenSCAD doesn't interpret these specifically, many developers still use them for clarity or for external tools. Changing them undermines that structure and can interfere with documentation generation or linting processes. Ideally, the formatter should preserve this style as-is.

**Workaround:** Adding a single-line comment (`//`) before the block comment avoids the issue. However, this is an undesirable fix, as it introduces a meaningless line of code solely to bypass a formatter bug and may disrupt tools that parse structured comments.

**Code:**

```
/**
* @file test.scad
* @brief Test file for OpenSCAD
* @author Your Name
*/

varx = 10;

cube([varx, varx, varx]);
```

**Resulting Error Output:**

```
[2025-05-18T21:31:25Z ERROR topiary_core] Failed idempotence check
[2025-05-18T21:31:25Z ERROR topiary_core] Diff < left / right > :
     /**
    < * @file test.scad
    < * @brief Test file for OpenSCAD
    < * @author Your Name
    < */
    >  * @file test.scad
    >  * @brief Test file for OpenSCAD
    >  * @author Your Name
    >  */

     varx = 10;

     cube([varx, varx, varx]);

[2025-05-18T21:31:25Z ERROR topiary] The formatter did not produce the same
    result when invoked twice (idempotence check).

    If this happened with the built-in query files, it is a bug. It would be
    helpful if you logged this error at
    https://github.com/tweag/topiary/issues/new?assignees=&labels=type%3A+bug&template=bug_report.md
```

### 2. Trailing Commas Introduced in Lists Cause OpenSCAD Syntax Errors (Critical)

Topiary appends commas to the last item in list arguments, which leads to syntax errors in OpenSCAD when passing them to modules.

**Original Code:**

```
// Test2.scad

module test_poly(points, paths) {
  polygon(
    points=points,
    paths=paths
  );
}

test_poly(
  [
    [0, 0],
    [10, 0],
    [10, 10],
    [0, 10]
  ],
  [
    [0, 1, 2, 3]
  ]
);
```

_Result:_ No errors.

**Post-Formatting Code:**

```
// Test2.scad

module test_poly(points, paths) {
  polygon(
    points=points,
    paths=paths
  );
}

test_poly(
  [
    [0, 0],
    [10, 0],
    [10, 10],
    [0, 10],
  ],
  [
    [0, 1, 2, 3],
  ],
);
```

_Result:_

```
Loaded design 'D:/repos/geodesic-dome/test2.scad'.
[ERROR: Parser error: syntax error in file test2.scad, line 20](https://github.com/Leathong/openscad-LSP/issues/20,D:/repos/geodesic-dome/test2.scad)
Execution aborted
```

**Fix (manual removal of trailing comma):**

```
// Test2.scad

module test_poly(points, paths) {
  polygon(
    points=points,
    paths=paths
  );
}

test_poly(
  [
    [0, 0],
    [10, 0],
    [10, 10],
    [0, 10],
  ],
  [
    [0, 1, 2, 3],
  ]   // <--- removed comma
);
```

_Result:_ No errors.

Leathong, 2025-05-19:
@CameronBrooks11 Thanks for the feedback. 
@mkatychev The second one is a new bug in topiary.

mkatychev, 2025-05-19:
Thanks for report. Looking at it on Monday

mkatychev, 2025-05-19:
@CameronBrooks11 [OpenSCAD playground](https://ochafik.com/openscad2/#H4sIAAAAAAAAE3WRUWvbMBDHv4q4MmjAa5sU2i6lDyljb6V2KPRByYpmS4ldWTLSKYnr+btP52RLxtiDZN3v/787c9dBI5yoPUw7EDmWG5kKXMMULhst2pWzwRQXPhcFJOBtcLmMVt7FrP+5cmtQGiTxkr1Ij5NBWZiFqW0RtGQY4VtjdXve2NKgTxhV8yPWLQxjJKysOac3ReR4OBgPjNwPw01gdL8wPZU/1iXM92Z+lbCr5SGTj/+JxsfwJNrfJyXGCZsk7Pqoju6hXyagpMDghqGAFh/t52BKayAqctdYh9+sqwVOvsZx+M0K/sbXA0YNfQKbUm5pCXGeNiC94rRk1OugsaTEokTrYIouyL1d/ony4NHW5QcRJbSXPe1Bkx3O1JfidpLT/tZ2O9vRv1JWtDRO/m77Qwe3Fp52+tPPu1BZ9ekZb1VmlVX0xTuVvT6qrOIqE63K1PxiVvGnt0hneyJasSX/qefpu0pfH0Ubs9+JVkvS43mPWSkRlRLBu4pHMieqbqjf0GtJp+Kx8m7ok4pdHM9KzSsOff8LBqACGbwCAAA=) has no issues rendering the second codeblock, what version of openscad are you running?

CameronBrooks11, 2025-05-19:
@mkatychev i was using latest release of OpenSCAD (2021.01)

mkatychev, 2025-05-19:
@CameronBrooks11 good catch on the version, the 2025 build handles this gracefully in the AST but 2021 does not, do you know if there's any other issues with trailing commas?
https://github.com/openscad/openscad/pull/3814 looks like this PR isn't in OS 2021.

https://github.com/tweag/topiary/pull/999 should have fixed the multiline issue.

Leathong, 2025-05-21:
> [@CameronBrooks11](https://github.com/CameronBrooks11) using [tweag/topiary#992](https://github.com/tweag/topiary/pull/992) should fix your issue

[image]
@mkatychev This still appends a comma after the last item in some list.

mkatychev, 2025-05-21:
> > [@CameronBrooks11](https://github.com/CameronBrooks11) using [tweag/topiary#992](https://github.com/tweag/topiary/pull/992) should fix your issue
> 
> [image]
> @mkatychev This still appends a comma after the last item in some list.

Yes the comma appending is intentional. The issue with trailing commas is specifically manifested in function calls: `foo([0,0],)` will throw an error and `bar([[0,0],])` will not.

Leathong, 2025-05-21:
@CameronBrooks11 You can downlaod the fixed version of topiary on the release page https://github.com/Leathong/openscad-LSP/releases

CameronBrooks11, 2025-05-22:
@Leathong Tested it out on my scripts and no more trailing comma error...looks like it still hangs on the impotency check but thats a much lesser problem. Thanks for all the support and a great project overall!