Leathong/openscad-LSP #52: Windows 10: Topiary not working, requires C compiler?

Zotteljedi, 2025-05-21:
After half a day of debugging and poking around in VSCode dialogs, I figured out that saving my changed OpenSCAD files doesn't work, because the formatter was switched to Topiary. Sorry for whining, but that was just a terrible user experience, completely disrupting my work flow.

Unfortunately, it still doesn't work for me, probably because I don't understand how Topiary works.

At first, I didn't want to install a wild binary into my system, completely cut-off from updates, so I decided to install Topiary via WinGet. This didn't work, because I got only notices about a detached Git head and a failed attempt to run **cl.exe**.

Then I opted for using the binary from the releases page, just to see that it fails due to a different missing compiler, **gcc.exe**.

So my assumption is that Topiary downloads a language specification from its Git repository and then attempts to compile a dynamic link library locally. Which means that I, as a user focused on creating 3D model files for printing, need to install and maintain a C compiler toolchain? Seems like a hell of a transitive dependency, if you ask me. :-/

For now I decided to disable formatting .scad files completely and to run clang-format manually before commiting my changes. If anyone is interested in how to do this: on Windows, edit %AppData%\Code\User\settings.json and add:

```
    "[scad]": {
        "editor.formatOnSave": false
    }
```

It's not a solution, but it's a way of getting my work done, and that's enough for me at the moment.

Leathong, 2025-05-21:
@Zotteljedi Formatting OpenSCAD is not yet included in the topiary official release. You can compile it locally from the newest source code.

Leathong, 2025-05-21:
You can use v1.2.5 if you didn't need formatting function.

Lenbok, 2025-05-21:
@Leathong I would also suggest following semantic versions for releases, removal of the clang formatting support is a breaking change that should have warranted a major version bump.

Leathong, 2025-05-22:
> [@Leathong](https://github.com/Leathong) I would also suggest following semantic versions for releases, removal of the clang formatting support is a breaking change that should have warranted a major version bump.

That is a good suggestion, I will release version 2.0.0 .

Zotteljedi, 2025-05-22:
I downloaded topiary-win.zip right now from the release entry v1.3.1, extracted it and get this:

```
C:\Users\Felix\Downloads\topiary-win>topiary.exe -vv format -lopenscad
[2025-05-22T07:03:05Z INFO  topiary_config::source] Could not find configuration file: C:\Users\Felix\AppData\Roaming\topiary\config.
[2025-05-22T07:03:05Z INFO  topiary_config::source] Using built-in configuration
[2025-05-22T07:03:05Z WARN  topiary::io] No query files found in any of the expected locations. Falling back to compile-time included files.
[2025-05-22T07:03:05Z INFO  topiary_config::language] openscad: Language Grammar not found, attempting to fetch and compile it
[2025-05-22T07:03:06Z INFO  topiary_config::language] openscad: Checking out https://github.com/openscad/tree-sitter-openscad.git acc196e969a169cadd8b7f8d9f81ff2d30e3e253
[2025-05-22T07:03:06Z INFO  topiary_config::language] openscad: Building grammar
[2025-05-22T07:03:06Z ERROR topiary] Error Fetching Language: Compilation error: Failed to execute the C compiler with the following command:
    "gcc.exe" "-O2" "-ffunction-sections" "-fdata-sections" "-m64" "-std=c11" "-I" "C:\\Users\\Felix\\AppData\\Local\\Temp\\.tmp1kPzeB\\openscad\\src" "-Wall" "-Werror=implicit-function-declaration" "-shared" "C:\\Users\\Felix\\AppData\\Local\\Temp\\.tmp1kPzeB\\openscad\\src\\parser.c" "-o" "C:\\Users\\Felix\\AppData\\Local\\topiary\\cache\\openscad\\acc196e969a169cadd8b7f8d9f81ff2d30e3e253.dll",
```

If that isn't how it should be invoked, please give me an example command line. The idea about `-lopenscad` came from VSCode's debug output, when clang-format was surprised about unrecognized parameters. ;-)

By the way, the UI in VSCode still shows "clang format executable path" as hint for scad-lsp.fmtExePath, and "clang format style, options: LLVM, GNU, Google, Chromium, Microsoft, Mozilla, WebKit, file" for scad-lsp.fmtStyle.

Ah, and the Changelog ends on v1.2.8 even though VSCode calls it v1.3.0 when hovering over the extension tile. Not sure what is going on here, I didn't do anything manually, it's all on auto-update.

Leathong, 2025-05-28:
@Zotteljedi Please update to v2.0.1, let me know if there is any bug.

closed 2025-05-28 by Leathong