Leathong/openscad-LSP #59: WASM build

lgarron, 2025-05-29:
I'm guessing that the main reason https://github.com/Leathong/openscad-support-vscode doesn't work out of the box is that it's complicated to install the platform-specific binary.

I think it would be sufficiently performant to ship a WASM binary bundled with the extension that can be used by default, which is well-supported by VS Code: https://code.visualstudio.com/blogs/2024/06/07/wasm-part2

There's also some prior art doing exactly this: https://www.osohq.com/post/building-vs-code-extension-with-rust-wasm-typescript

The WASM could be invoked by the runtime in the extension, while also being runnable directly using a tool like Wasmer: https://wasmer.io/

I started taking a look at compiling this project to WASM, but I ran into immediate issues because this project relies on C code:

```
cargo:warning=In file included from /Users/lgarron/.cache/cargo/registry/src/index.crates.io-1949cf8c6b5b557f/tree-sitter-0.25.5/src/lib.c:1:
  cargo:warning=In file included from /Users/lgarron/.cache/cargo/registry/src/index.crates.io-1949cf8c6b5b557f/tree-sitter-0.25.5/src/./alloc.c:1:
  cargo:warning=/Users/lgarron/.cache/cargo/registry/src/index.crates.io-1949cf8c6b5b557f/tree-sitter-0.25.5/src/./alloc.h:9:10: fatal error: 'stdio.h' file not found
  cargo:warning=    9 | #include <stdio.h>
  cargo:warning=      |          ^~~~~~~~~
  cargo:warning=1 error generated.
```

It should definitely still be possible to compile this to WASM, but I think it requires a custom `build.rs` file. Or perhaps a tool like `wasm-pack` could also work.

I'm not sure how far I'll get by poking at it, but I figured I'd at least file an issue.

Leathong, 2025-05-30:
WASM might not be a good solution because we need to load code from local files, but WASM cannot do this.

lgarron, 2025-05-30:
> WASM might not be a good solution because we need to load code from local files, but WASM cannot do this.

That sounds… incorrect to me?

It's possible to give access to WASM in a few ways. WASM supports arbitrary bindings to JS and there is standards work to implement specifically a file system (https://github.com/WebAssembly/wasi-filesystem) as well as a full POSIX API (https://wasix.org/).

My understanding is that `wasi_snapshot_preview1` is the most reusable form of this right now. If you're okay with having an extension dependency in this case, Microsoft has an official extension and `npm` package that is meant to hook up the bindings with minimal boilerplate: https://marketplace.visualstudio.com/items?itemName=ms-vscode.wasm-wasi-core